bigtreetech/BIGTREETECH-TFT35-V2.0 #3: Mesh bed levelling problem

StoneFist85, 2019-09-19:
Hi, I just recently purchased SKR 1.3 and TFT 35 V2.0 Finally today started working on it and changed it from MKS GenL  with stock Ender 3 display to this one and now I can not enable mesh bed levelling. This was all working fine on MKS but not sure now if this is problem with Marlin 2.0 on board or something with display itself. When I press Levelling on display it only homes all axes and does nothing more afterwards.

Zuru1, 2019-09-19:
Did you get the latest firmware and enabled auto bed levelling?
Any way at the moment its only manual bed levelling which is available via the display. The rest of the available ABL types need to be done via prontoface or similar while connected to a computer.

StoneFist85, 2019-09-20:
Yes, I got latest firmware for both board and tft screen. There is option on screen for levelling but after pressing it all it is happening is homing all axis. I've double checked Marlin 2.0 and I got it setup the same way as I had it on 1.1.9 with MKS where it works but here is no joy.

turnali, 2019-11-28:
have you fixed the problem? i am dealing with the same think but on MKS board with TFT35 V3

d3vilMod, 2020-02-05:
same problem of @StoneFist85 
enabled the mesh bed leveling but when press the ABL the printer do homing and nothing else.

turnali, 2020-02-05:
> same problem of @StoneFist85
> enabled the mesh bed leveling but when press the ABL the printer do homing and nothing else.

i was able to fix the problem with the latest version of Marlin. Just update your Marlin software and reinstall it to your board. That should fix it

d3vilMod, 2020-02-06:
> > same problem of @StoneFist85
> > enabled the mesh bed leveling but when press the ABL the printer do homing and nothing else.
> 
> i was able to fix the problem with the latest version of Marlin. Just update your Marlin software and reinstall it to your board. That should fix it

in the tft firmware did you disable the "auto_bed_leveling"??? i think that it can go in conflict.

turnali, 2020-02-06:
> in the tft firmware did you disable the "auto_bed_leveling"??? i think that it can go in conflict.

i didnt change anything in the tft firmware. i just downloaded the latest version of it with the icons and the fonts folder and installed it to the tft display. just make sure about the right baud rate

StoneFist85, 2020-02-10:
Well tbh I kind of left it as it was and used old screen. Now just got myself BL-Touch so will probably try it again with ABL on.

robstermn, 2020-02-17:
Any progress with this issue, I have the same problem.  

[image]

bigbosshogg, 2020-02-17:
I'm having the same issues as robstermn, as well as an error when trying to change the LCD color. That returns and Unknown Command "M150 R0 U0 B255 P255" ok error message.

transmatico, 2020-02-18:
Me too. have the same issue { Failed to enable Bed Leveling Echo : Bed Leveling OFF Echo: Fade Height OFF ok }

bigbosshogg, 2020-02-18:
Found out I can edit the configuration.h file and disable ABL. Seems it is on by default, but I wanted to try mesh leveling over my BL Touch. I swear it isn’t acting right. Now I am not sure what to do about the M150 error I’m getting. I want that pretty light!!! It worked before so I don’t know what happened.

schaudi, 2020-03-05:
The Error on first boot accures, cause there is no saved mesh to load. U can do Mesh Bed Leveling in LCD Mode, if u uncomment #define MESH_BED_LEVELING (and of course set every related setting as u need)  and #define LCD_BED_LEVELING in the Firmware of ur Marlin Board. As soon as u have a saved Mesh, the Error Message won't show up again in TFT Mode.

Angel215i, 2020-03-06:
Ive done mesh bed leveling so many times with the tft35 v3.0 nothing works i need help ive tried what Schaudi mentioned nothing again

questionablemechanics, 2020-09-06:
Having similar issues with my TFT24 on an SKR 1.4v { Failed to enable Bed Leveling Echo : Bed Leveling OFF Echo: Fade Height OFF ok }