bigtreetech/BIGTREETECH-TFT35-V2.0 #4: LCD comment in Marlin 2.0

mpieroway, 2019-10-03:
I have The BIGTREETECH TFT35 V2.0 and have it connected to an MKS SBASE v1.3. I am looking for the proper LCD to comment in in Marlin 2.0 in order to have the LCD connect to the control board. Regardless of the option I comment in the LCD continues to display "no printer attached"

Thank you

hutchinsane, 2019-10-04:
The display connects to one of the UART ports on the motherboard and it communicates with the 3D printer via serial and gcode, there's no LCD option you have to uncomment in Marlin

mpieroway, 2019-10-14:
The display is connected to the serial port on the board where the MKS TFT would connect with serial connector. However, the Tx and Rx have to be reversed when connecting to the board. Once the second serial port was commented in in Marlin, everything works.

Cheers

closed 2019-10-14 by mpieroway