bigtreetech/BIGTREETECH-TFT35-V2.0 #7: TFT35 V2.0 + ESP01S Wifi

Doom6666, 2019-10-20:
Hi, I have a TFT35 V2.0 with ESP01S attached. I am connected to my home WiFi. However, it is not working to transfer files to the SD card located in TFT35. Can anyone help? Thanks

grego1981, 2019-10-21:
As mentioned in a another [issue](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/43), this is not implemented yet. How did you connect it to the WiFi?

Doom6666, 2019-10-21:
> Jak je uvedeno v jiném [čísle](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/43) , toto ještě není implementováno. Jak jste se připojili k WiFi?

I uploaded ESP3D firmware to ESP01S and inserted ESP3-WebUI with configuration as CLIENT ... inserted into slot on TFT35 V2.0. I can control the printer via WiFi, but I still cannot upload files to SD.

luc-github, 2019-10-21:
if your TFT is connected using serial (touch mode) ESP3D has no way to access the TFT SD card reader, it can only access to printer SD card if any

dineshvyas, 2019-10-23:
> > Jak je uvedeno v jiném [čísle](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/43) , toto ještě není implementováno. Jak jste se připojili k WiFi?
> 
> I uploaded ESP3D firmware to ESP01S and inserted ESP3-WebUI with configuration as CLIENT ... inserted into slot on TFT35 V2.0. I can control the printer via WiFi, but I still cannot upload files to SD. I have V2.0.23 firmware on TFT35 V2.0...

Can you post step by step instructions?

TimboNZ76, 2019-11-25:
Any progress on this situation ie upgrades etc?

acidcoolxxx, 2019-12-05:
Have the same issue. For now I use the Onboard option. Change #define SDCARD_CONNECTION LCD to #define SDCARD_CONNECTION ONBOARD and you have at least access to it over wifi. Use **M22** to unmount the SDcard so it will be displayed again on your PC.

dineshvyas, 2019-12-18:
It is easier than I could believe, configured with hit and try. Working! wifi with SKR_PRO_1.1.

epicurean70, 2020-01-15:
IS esp01s supported on the tft35 v2 now?

Mikomanus, 2020-02-13:
yep it is. But i still can't get the SD card working over wifi.

luc-github, 2020-02-13:
Local tft SD card is not supported in ESP3D

baccula, 2020-07-28:
Any plans to implement local SD access? My printer board doesn't have SD so I have to use the TFT slot.