bigtreetech/BIGTREETECH-TFT35-V2.0 #11: Marlin 2.0 with TFT35 V2.0 and SKR V1.3

Tothyr, 2019-12-03:
I have an issue with the configuration.h and configuration_adv.h.
I dont know what parameters change or modify.
Anyone can help me? Thanks for advance!

mpieroway, 2019-12-03:
What's the issue? If it is getting the serial port to power the display, you have to ensure both serials are commented in, in the configuration.h

Tothyr, 2019-12-03:
Is "ON", but there is not responding the commands and says "not printer attached".

mpieroway, 2019-12-03:
Have you updated the firmware on the display?

Tothyr, 2019-12-03:
No, How? and what firmware I have to install?
Thanks for the patience and dedication!

Tothyr, 2019-12-03:
I dont have to touch anything in the Marlin configuration?

mpieroway, 2019-12-03:
If both serial ports are commented in in marlin (if using marlin 2.0) then nothing else needs to be changed in marlin. 

Firmware update of the display is done through the sd card. Download the most recent copy from the BIGTREETECH github for your display. There should be directions in the documentation on github on the exact process for the update

Tothyr, 2019-12-03:
Thanks a lot! This weekend I try this.

DaVincIan, 2019-12-17:
Tothyr

I had the same problem, but solved it.
1) You do not change any configuration setting(s) in Configuration.h to Configuration-adv.h that relates to screen/lcd/touch.

In my case I had in Configuration.h (about line 125)
#define BAUDRATE 250000
and I changed it to
#define BAUDRATE 115200

I also have
#define SERIAL_PORT -1
and
#define SERIAL_PORT_2 1

The TFT firmare updates are here (and how to do it):
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Hope you come right.

bhusum, 2019-12-28:
Having the same issue w/ the "no printer attached" message.  Hardware is the SKR v1.3 and TFT35-V2.0. This is a conversion project that is updating a CR-10 w/ the above hardware.  Basically I set the machine up using the stock CR-10 screen and configured Marlin-bugfix-2.0.x to test all the functionality with the SKR.  That process worked and the machine functions fine with the stock LCD.  Once I had everything working the next step was upgrade to the touch screen.  This is where I'm having problems.  

On the TFT35-V2.0 I have successfully updated (as confirmed on the TFT info screen) to the latest firmware:  
BIQU_TFT35_APP1_V2.0.25.bin

Set the:  
#define SERIAL_PORT -1  
#define SERIAL_PORT_2 0
#define BAUDRATE 115200

Connected (and checked the continuity) of the black (5) wire ribbon cable to the SKR V1.3 pins marked TFT on the control board and plugged the connector in to the RS232 plug on the TFT.  When I do that TFT powers up and I am able to move the printer axis, home, adjust fan speed but not see any temp info on the screen and there is the "No printer attached!" message being displayed.  It is the same for both the classic and the unified versions of the firmware. 

I've tried various things mentioned above and in other posts but can't seem to get the printer connected.  When I change the serial port definitions as mentioned to:

#define SERIAL_PORT -1  
#define SERIAL_PORT_2 1
#define BAUDRATE 115200  (or 250000)

I am unable to move the the printer axis at all.  I'm not sure if I have a something in the 'LCD Controller" section defined in Configuration.h or Configuration_adv that is causing an issue.  Feels like I might have something configured when I was using the stock CR-10 for testing that may be a problem? When using the TFT's do you comment out anything that has to do with a LCD controllers in Configuration.h?  Attached are the configuration files that I am using at the moment that seem to allow me to move the axis but not connect to the printer if that helps.

 [ConfigurationFiles.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V2.0/files/4007410/ConfigurationFiles.zip)

DaVincIan, 2019-12-30:
Bhusum

I have the SKR pro. SKR V1.3 and SKR Pro configurations are different.

I have problems with mt TFT35 V2 as well. I did the hex upgrade. After uploading the "BIQU_TFT35_APP1_V2.0.25.bin" to the TFT, the filename does not get renamed to "BIQU_TFT35_APP1_V2.0.25.CUR" as when I upload the "firmware.bin". Do you experience the same?

Also, the TFT text is "scrambled" with this TFT firmware version. It looks as if the incorrect font is selected. When I change the language, the Chinese characters are correct.

bhusum, 2019-12-31:
DaVincIan-
So I did not apply the BIQU_TFT35_V2.0_bootloader.hex fix to the TFT35-V2.0
when I received it if that is what you are referring to.  For some reason I
was thinking that fix was to address the issue with being able to
upgrade the firmware though the SD card that some of the older firmware was
having.  I was assuming it was working because the version changed on the
info screen on the TFT.  May not be a correct assumption?

Not sure which firmware came with your board, but TFT35-V2.0 that I have
was ordered from Biqu Equipment on 12/5/19 and arrived 12/26/19 and I
believe that it had BIQU_TFT35_APP1_V2.0.24.2.bin firmware installed.
 Went back and copied over a fresh BIQU_TFT35_APP1_V2.0.25.bin and the
TFT35 folder and reinstalled the firmware to see what would happen.  From
what I can tell it looked like it updated the fonts and bmp fine, but the
BIQU_TFT35_APP1_V2.0.24.2.bin did not change to
BIQU_TFT35_APP1_V2.0.24.2.CUR.  When I was working on the SKR board it
always changed the .bin to a .CUR once it was rebooted.  What is
interesting though is that it added the .CUR extension to the file folder
that contained the fonts and bmp.  TFT35.CUR is what it turned into after
rebooting it.  Is that what you were seeing as well?  Is that what the hex
update is supposed to fix?

As far as the TFT text goes, the only issue that I ever had with the
characters not showing up right was when I first tried to flash the
firmware using the .bin file along with the font and bmp folders instead of
the .bin and the TFT35 folder.  Flashing using the .bin, font folder, & bmp
folder made all the text scrambled which was my mistake.  Not sure where to
go from here.

DaVincIan, 2020-01-01:
Bhusum

I was busy writing you a long explanation of my experience when I decided to read your post again. It provided me with a clue as to what I did wrong.

Like you, my TFT35 V2.0 arrived with FW BIQU_TFT35_APP1_V2.0.**22**.bin. However I could not print from file using this version. It kept asking for cold extrude and warned that the filament sensor triggered.

So I did the "TFT35_V2 Bootloader fix" as per the instruction.

I could flash version BIQU_TFT35_APP1_V2.0.25.bin of the firmware, but my font was scrambled, because I placed the bmp and font directories with its content in the SD card root, instead of in the directory TFT35.

When I corrected my mistake, SD:\BIQU_TFT35_APP1_V2.0.25.bin, SD:\TFT35\bmp and SD:\TFT35\font, the bitmaps and fonts installed and directory TFT35 was changed to TFT35.CUR.
BIQU_TFT35_APP1_V2.0.25.bin installed, but no changes to the file name. 

Now everything looks good. I have not tried printing from it yet. I did manage a print form file (TFT SD) with the text scrambled, so I believe the FW is working.

I'm not sure what the hex upload fixes.

Thanks for your help.
Ian

bhusum, 2020-01-01:
Hi Ian-
Glad you got that part worked out.  Does your TFT still give you the "No
printer attached"?  Let me know if you can print from the TFT's SD.  I'm
not able to do that.  I've always just kept the buad rate at 115200 in
Marlin and made sure that it was selected in the TFT's settings menu.
Rebooting everything and rechecking anytime I made any changes.  Still no
luck.  When you said you were able to print from the LCD's SD when the text
was still scrambled, what were you using for the SD connection type in
config.adv.h?  Think it is around line 1154 or so.

 #define SDCARD_CONNECTION ONBOARD      // 'LCD', 'ONBOARD', 'CUSTOM_CABLE'

I've tried both the ONBOARD and LCD options here and it doesn't seem to
make a difference.  Read in another post somewhere that LCD was the correct
setting if you wanted to print from the TFT's SD.  Don't know though.

So this may be slightly off topic and best reserved from another post, but
I have also been trying to get a TFT24-V.1.1 working as well.  That is the
smaller screen with the encoder that allows you to switch between Marlin
mode and touchscreen mode.  Running into similar problems when trying to
get the touch screen functionality to work.  I was able to get the
TFT24-V1.1 working correctly in Marlin mode when defining:

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1    (also works with them reversed.  -1 on P0 and 0
on P2)

This allows me to print from the SD card on the SKR and works exactly like
the old CR-10 with the stock LCD.  When running the TFT24 in marlin mode
you do need two ribbon cables connected to the EXP1 and EXP2 plugs on the
SKR.  When I disconnect the EXP2 cable from the board I lose the ability to
interface with the SKR.  The screen is still lit up but can't scroll though
the menus or print anything.  It makes me think that there is something
like that going on with the touchscreens.  I don' really know though.

Brent

DaVincIan, 2020-01-02:
Brent
_Does my TFT still give you the "No printer attached"?_ No. It shows for about 10s to 20s when I turn the printer on, or perform a reset.

My Marlin is set to: #define SDCARD_CONNECTION ONBOARD. But I can only print from the TFT screen SD card, not the controller SD card. When I select "Print" / "OnboardSD" on the TFT,  I get and error: "Read onboard SD card error!"

For the SKR Pro the serial ports are defined as:
#define SERIAL_PORT -1
#define SERIAL_PORT_2 1

I know it is different for the SKR1.3

I'm planning to switch to the 12864 LCD. From what I've read it is more compatible with Marlin functionality. I hope easier to setup...

Ian

bhusum, 2020-01-02:
Ian-
I'm thinking along those same lines WRT to the LCD.  Using these TFT's are
starting to feel like a complicated solution to a simple problem.  I'm
going to give it one more chance and check it on one of the new SKR V1.4
boards that is suppose to show up here in a few days.  Maybe something will
be different on that one.  Thanks for you help.  B.

DaVincIan, 2020-01-03:
Brent

I'm puzzled that you struggle to get your SKR V1.3 to work. From the online information I came across, it seemed like an established controller to setup and configure. 

Look at Edward Braiman's series of YouTube videos...they are quite good: https://www.youtube.com/watch?v=45NMUl7GnDM
I see he now has a series on SKR V1.4 as well.

If you want I can have a look at your two configuration files. May I pick up something you missed.

Ian

bhusum, 2020-01-03:
Hi Ian-
I actually got the SKR v1.3 and both the TFT35 V2.0 and the TFT24 V1.1
working last night.  My SK1.4 Turbo board arrived and I was going through
the process of comparing and transferring the pertinent settings from the
SKR V1.3 configuration files to the ones downloaded from BTT's Github page
for the new SKR 1.4 Turbo.  I basically commented out a lot of the advanced
features from my SKR v1.3 configuration and tried to keep it as close to
the SKR V1.4T configuration as possible aside from the definitions that had
to do with the physical hardware, drivers, etc.  At a high level the
definitions that I commented out to get the thing working had to do with
advanced park, persistent EEPROM, BL-Touch, and a couple other minor
features that at the time didn't seem that important.  I really didn't make
any changes to either serial port definitions, baud rate, SD card (onboard
vs LCD) etc.  I'm thinking now that there had to be some setting in my
original configuration that was causing a conflict somewhere.  Maybe the
I really need to go back and figure out which one of those definitions (or
combination of) was causing the problem now and see if I can recreate it.

So for now I think everything works with the TFT's and SKR v1.3 w/o having
some of those extra features enabled.   Able to print from either TFT35
V2.0's SD and the SKR's onboard SD.  Same with the TFT24 V1.1's the USB on
that one as well.   The "no printer attached" message disappears once
Marlin gets up and running now as it should.  I appreciate you offering to
take a look at my configurations and I would be more than happy to send
them to you if you want to take a look at them.  Thanks for the help.

Brent

PS.
These are files from the BBT Github page for the SK1.4 Turbo that I used to
compare w/ my configuration for the SKR v1.3 using Notepad++ when I was
trying to get the TFT's to work.  *I did not compile and upload this to the
SKR v1.3 for anyone reading this!*  I just used it to check the definitions
and settings to make sure they were similar.

https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-2.0.x-SKR-V1.4-Turbo

Interplain, 2020-01-11:
Hi, looking for some help. Is it possible to get this screen to show the current Z offset position like the old Anet Screen did. How are you supposed to setup Offset if you can't see where it is currently positioned. Is there a firmware that will show this.

PhilMaddox, 2020-05-14:
The big issue i see here is people thinking the TFT35 V2.0 is like other LCD screens they can configure in Marlin and connect. Its not. Its a standalone smart device that acts like a laptop or raspi connected to the USB serial port on the SKR1.3.  All it requires is that its connected correctly from "TFT" on SKR1.3  to RS232 on TFT35-V2.0  and that the baud rate set in Marlin ( most use 115200) is also set on the screen settings. 
[image]
Any issues with fonts / layouts / language are to do with the TFT35-V2.0 firmware and the config.ini which should be present in the root of the SD when firmware is flashed

RavenWarrior, 2020-06-01:
I’m having the same issue... the cable is in correct. Loose reset pin nearest to the sd port.
Serial port -1
Serial port 2 0
Tried them the other way, tried -1 and 1
Baud rates match.
Whatever I’ve found in all the forums and videos on YouTube. I’ve tried. Even talking to some guy in Norway on WhatsApp who got his to work with old 2.0.25 version, he showed me how the new version doesn’t work for him but the old does. And yet none of them work on mine. Some scramble the images. Some are ok. But none get rid of the ‘no printer attached’
I’m using the btt skr 1.4 turbo with tft 35 v2.0.
I’m not expecting a solution, I’m assuming it’s hardware issues. But I’m just here to say it isn’t just because we noobs got the baud rate wrong. Or ports wrong...
Could it be that the issue is in some of the marlin versions and not others.

bhusum, 2020-06-01:
Hey Andrew-
Sorry to hear your are having trouble.  I went through the same issues back
when I started that first post.  Wanted to reply to this because I recently
upgraded from the SKR1.3 to the SKR1.4 Turbo recently and am also using the
TFT 35 V2.0 and am not having any issues at the moment.  When I was
originally having the "no printer attached" problem I think I was trying to
configure too many things at once and I probably had something crossed up
that I wasn't aware of due to my inexperience.  The way I got it to work
was to start with a fresh copy of Marlin and then started configuring
things one by one and tried to keep it as simple as possible.

Anyway I'm running a modified CR-10, but if you would like me to send you
my configuration files so that you can take a look at them or some pictures
of my board I'd be happy to do that if you think it would help.  Right now
I'm running a Marlin 2.0.5.3 on the SKRv1.4Turbo and
a BIQU_TFT35_APP1_V2.0.25.1 that has been customized for my machine on the
TFT 35 V2.0.  Not sure if you use Notepad++, but it has a great plugin that
allows you to do a document compare and see how your configuration files
compares to someone else's.

Brent

RavenWarrior, 2020-06-02:
> Hey Andrew- Sorry to hear your are having trouble. I went through the same issues back when I started that first post. Wanted to reply to this because I recently upgraded from the SKR1.3 to the [...]

Hey, I’ll give the fresh marlin, I’m currently using 2.0.x.... I’ll get back to you in a sec

RavenWarrior, 2020-06-02:
> Hey Andrew- Sorry to hear your are having trouble. I went through the same issues back when I started that first post. Wanted to reply to this because I recently upgraded from the SKR1.3 to the [...]

Ok, so I did a fresh marlin install.
Changed serial port to 0
And serial port 2 to -1
Baud 115200.
Set the same on the screen.
Changed environment to lpc1769
Board 1.4skr.
They’re the only things I changed.
Still no printer attached...
I don’t know what else to do.
Ima check out the notepad++ like you said.

Zuru1, 2020-06-02:
Have you tried setting the first serial port to -1 and the second port to 1?

RavenWarrior, 2020-06-02:
> Have you tried setting the first serial port to -1 and the second port to 1?

I’ll try it, I tried it on my original marlin but I suppose it’s worth a try on the fresh marlin

RavenWarrior, 2020-06-02:
> Have you tried setting the first serial port to -1 and the second port to 1?

No change...

RavenWarrior, 2020-06-02:
Is there a way of comparing in vscode?
I can’t get the notepad++ to compare because it doesn’t have the plugin manager...
Tried uploading the github compare tool for it but couldn’t get that to work.

bhusum, 2020-06-02:
Hey Andrew-
Ok here are a few things you can try if you are interested. Attached is a
zip file containing some files that I am currently using for my
modified CR-10 with the SKRv1.4 Turbo.

SKR V1.4 Turbo Files Folder:
This folder contains my configuration.h and configuration_adv.h if you want
to go through them.  These were based on Marlin 2.0.5.3.  The folder also
contains a firmware.bin file that I just created if you want o try flashing
your board with it to see if anything changes. Might save you some time
instead of trying to start from scratch.

TFT35 File Folder:
This folder contains my configuration.h for the the TFT if  you want to
look at that for some reason.  It is basically customized for my machine
with some presets for material temps, probe points etc.  Also in this
folder is a fresh binary that I just built for the TFT that you can try
flashing the board to see if it works. You probably already know this, but
I'll mention it because it messed me up early on.  Make sure that the
BIQU_TFT35_APP1_V2.0.25.1.bin and the TFT35 Folder are in the root
directory on your SD or it won't work.  If the update works right there
will be a folder called TFT.CUR on the SD card after the update when you
pull it out an look at it on your computer.  Let me know how it goes.
 Brent

bhusum, 2020-06-02:
Not sure about the compare in VS Code. Haven't really been using it that
long so I don't know my way around it yet.  B.

RavenWarrior, 2020-06-02:
WHERE DO I FIND THE ATTACHMENTS?
sorry caps lock

bhusum, 2020-06-02:
Sorry about that...tried to reply straight from my email with an attach and apparently that does not work or is not allowed.  See if this zip file posts.

[060220_ConfigFiles.zip](https://github.com/bigtreetech/BIGTREETECH-TFT35-V2.0/files/4719718/060220_ConfigFiles.zip)

RavenWarrior, 2020-06-02:
> Sorry about that...tried to reply straight from my email with an attach and apparently that does not work or is not allowed.  See if this zip file posts.
> 060220_ConfigFiles.zip

ha, I was busy replying to your last post. I found it further up the page

bhusum, 2020-06-02:
No worries.  See if any of that stuff works.  Thought if you flashed that
binary on your machine even though your printer is probably different it
may give you an idea if your hardware is working. B.

RavenWarrior, 2020-06-02:
> No worries.  See if any of that stuff works.  Thought if you flashed that
> binary on your machine even though your printer is probably different it
> may give you an idea if your hardware is working. B.
> …
>  Sorry about that...tried to reply straight from my email with an attach
>  and apparently that does not work or is not allowed. See if this zip file
>  posts.
>  060220_ConfigFiles.zip
> 
>  ha, I was busy replying to your last post. I found it further up the page
> 
>  <#11 (comment)>,

thanks for the efforts B,  I tried it both your board and lcd firmware. no change... ill just have to wait for Biqus' reply.
they did get in touch but just suggested all the stuff you guys have tried.
I told them everything i'd attempted but I don't know if they'll have anything else to suggest, unless I had enough grounds for them to send a new board/tft out.
might be a dumb question but im running my board from a be quiet pc power supply, could that be an issue?

bhusum, 2020-06-02:
Not sure.  Pretty sure those pc switching power supplies need a load on
them if you are using a converted one for a printer.  I used to use 3 or
4,  25 watt sand resistors ( 100 ohm maybe?) to put a load on them to get
them to work.  Check Google but that might be your issue.  I'm running a
hlg-320h-24b for a power supply for the everything but the heater bed.
Have an SSR and AC heater bed for that.  If you have something else give it
a try.

RavenWarrior, 2020-06-02:
> Not sure.  Pretty sure those pc switching power supplies need a load on
> them if you are using a converted one for a printer.  I used to use 3 or
> 4,  25 watt sand resistors ( 100 ohm maybe?) to put a [...]
My printer works fine, its just the lcd.
ive got an external mosfet for the heat bed. I just twisted the sense wire in the power supply with another, I forget which one. Ive even got sensorless homing on the z, which lifts with a belt, driven with a worm drive setup.
just wanted a nice touch screen to finish the aesthetics lol
I appreciate your time B.

bhusum, 2020-06-02:
OK.  One other thing to check.  I was literally just printing something on
my machine right now and wiggled the wires on the ribbon cable connecting
to the board ever so slightly and the the LCD lost connection and gave me a
no printer error.  Did you check the actual ribbon cable itself to make
sure that all the conductors were enact?  Anyway hope you can get it worked
out.  B.

RavenWarrior, 2020-06-03:
> OK. One other thing to check. I was literally just printing something on my machine right now and wiggled the wires on the ribbon cable connecting to the board ever so slightly and the the LCD lost [...]

I really hoped for a moment that it was this.
I checked with the multimeter. They all beep (sad face).
Good call though. It all gets a bit deep when really it could be something as simple as a broken wire .

RavenWarrior, 2020-06-03:
I’ve just noticed while uploading my marlin, that it’s compiling the ftdi eve touchscreen. Even though its uncommented.
It’s compiling from .pio\build\Lpc1769\src\src\Lcd\extui\lib\ftdi_touch_ui\screens\ and all that stuff in there.
It’s also compiling other screen stuff like ultra lcd
Malyan lcd.
Shouldn’t be the problem because I’ve tried your firmware directly onto sd and some other guys from reddit.
But I’m wondering now if vscode is including some library that’s interfering with the screen communication

PhilMaddox, 2020-06-03:
> I’ve just noticed while uploading my marlin, that it’s compiling the ftdi eve touchscreen. Even though its uncommented.
> It’s compiling from [...]

WHen i complie for SKR1.3 ( Marlin 2.0.5.3) i see all those items scroll past too but it doesnt effect the end result. In fact the latest firmware i compiled uses the standard LCD 2004 and it works along side the TFT35 V2 fine

PhilMaddox, 2020-06-03:
Maybe a simple problem but have you double checked that the TX & RX are crossed i,e, TX on the TFT goes to RX on the SKR1.4.  That has caught me out on other set ups before

RavenWarrior, 2020-06-03:
> Maybe a simple problem but have you double checked that the TX & RX are crossed i,e, TX on the TFT goes to RX on the SKR1.4. That has caught me out on other set ups before

Omgg what!!! I just just checked the diagrams and the screen goes power/gnd/rx/tx/rst 
The 1.4 board (you’re correct) power/gnd/tx/rx/rst...
Why would they do that!!!
Ima try it now.....
One sec
... ... ... .
Nope...
Still displaying ok but still no printer attached
Should it change the firmware bin to cur?
It changes the tft35 folder to cur. But not the firmware

RavenWarrior, 2020-06-03:
The screen says uart speed... is that just baud rate or is there another setting in config.h that should be set, like card spi speed or sd card enable check and retry?
I don’t have them commented

PhilMaddox, 2020-06-03:
Yes baud rate & UART speed refer to the same thing in this case. Both should be 115200 .

In Marlin the SD card enable ( uncomment) refers to screens that use EXT 1 & EXT 2 ports on the SKR 1.3/4. The TFT35V2 handles it's SD card slot internally so no changes are need in Marlin

RavenWarrior, 2020-06-03:
Damn. Well biqu have been in touch. They initially told me to try all what you guys have tried. And then they asked for a video... just a video. So I filmed pretty much every scenario and told them all what I’d tried. See what happens

PhilMaddox, 2020-06-03:
To be clear the TX & RX have to be crossed to work! TX on SKR 1.4 should be connected to RX on TFT 35V2. I've noted the 1.4 has the pins 18O rotated compared to 1.3 . On the 1.4 the RST is nearest the SD card slot whereas on the 1.3 it's furthest away!

RavenWarrior, 2020-06-03:
Ok. Well It was right then. But just to confirm, it doesn’t work with them the other way

RavenWarrior, 2020-06-03:
Here’s a different question in relation to the tft...
If your tft won’t connect, and yet you want to print over usb cable. Which port do you have to comment on for it to appear in cura?

maker-DJTodd, 2020-06-03:
Anyone have luck resolving this?

I am having same issue tft35 v2.0 with skr1.3. Powers on fine but no printer attached. 

I’ve tried both the latest and one release previous Tft firmware, and every combination of baud rate and serial 0 or 1. 

Also tried a fresh marlin to be safe with the same result.

PhilMaddox, 2020-06-04:
> Here’s a different question in relation to the tft...
> If your tft won’t connect, and yet you want to print over usb cable. Which port do you have to comment on for it to appear in cura?
> ...
> Never mind. Just take the sd out

Just plug in the USB cable . Take a look in device manager and see which comm port is allocated then use that in Cura

RavenWarrior, 2020-06-04:
I’ve been in touch with biqu...
They asked me if I’d tried all the solutions on this open issue.
I told them I’d tried everything.
They then asked for a video.
Of what they didn’t specify, so I filmed everything.
They then replied, asking me to check for a short on rx tx.... just a sec...
This is the message;
Hello,
Can you check if RX, TX and GND are shorted? 
If there is no short circuit, change a touch line. 
If there is still no effect, and the display chip is abnormally hot. It may be a chip problem.

Yours sincerely,
BTT customer service

I asked for a video,
They said 
Hello, no need to.

I’m assuming I’m to put my ground cable on gnd
And line on rx and then tx.
If there’s a short then I have no idea what a touch line is.
So I’ve asked a few more questions just to be sure.
Oh and I’m assuming I’m checking the screen and not the board, but I’ll check both.

RavenWarrior, 2020-06-04:
[image]

This is what they sent just now... and said you check those for continuity.

RavenWarrior, 2020-06-04:
Should there be continuity between pins on the tft?
I’m getting continuity on all.
Only the power and reset don’t have continuity together

watzkej, 2020-06-05:
I'm having the exact problem as @RavenWarrior.  I have a SKR1.4 Turbo and the TFT35 v2.  I've tried the suggestions and also cannot get anything to work.  I'm not sure I understand what support is asking you for.  I'll have to pull my board out and look at the bottom as well.  I've written to support but haven't gotten a response yet so I'm hoping I might get some progress here.

watzkej, 2020-06-05:
I'm thinking there's simply something wrong with the TFT35 detecting the printer and getting feedback.  Ignoring the "No Printer Connected" message, I tried to do a few things on the printer via the TFT35.  I can preheat for PETG then cooldown, turn fans on and off, move each axis around, and even extrude filament.  It seems like the printer receives commands but isn't able to give feedback to the LCD.  I was going to attempt to print from the SD card on the LCD but it can't open the SD card even though I was able to flash the firmware with that same card.  I have my original Ender3 Pro LCD still hooked up so I'm able to see the status of the preheat to know that it's actually setting the hotend to 240 as I sent from the TFT35.

@RavenWarrior, are you able to get the printer to at least listen to commands?

maker-DJTodd, 2020-06-05:
Ok, so mine is now working. There were a few things in the BTT video that I did not see anyone else mention. The video referenced is https://www.youtube.com/watch?v=oaXfXkPYHpw&t=8s

The relevant changes were 
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define BAUDRATE 115200
#define SDSUPPORT (just uncommented)

Works like a champ now. 
Confirmed with latest tft firmware, both the legacy and unified menu versions.

Dunno if that helps anyone else, but worth a shot. :)

watzkej, 2020-06-05:
That didn't make any difference for me aside from having to set the TFT to 115200 in order to send commands.  It still seems like it isn't able to get feedback to know the printer is there.

You have a SKR1.3 and I have a SKR1.4 Turbo so it's possible the serial ports might need to be a little different.  They don't appear to have an SKR1.4 video showing FW config settings unfortunately.  It feels like I'm close since I can at least send commands.

bhusum, 2020-06-05:
I have run both the SKR V1.3 and more recently the SKR V1.4 Turbo & TFT35
V2.0 with the serial ports definitions below and they both worked.

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1

This may be a total shot in the dark, but how are you defining your
//#define SDCARD_CONNECTION LCD in the configuration_adv.h?  Think it is
around line 1207.   This line is commented out in my file and it seems to
work but I remember having problems when I was changing this to what I
thought should be defined as LCD.  B.

RavenWarrior, 2020-06-05:
I’ll try moving some axis with it tonight, I’m pretty sure I’ve tried before but I can try again.
as I said I’ve got all the settings right.
Port and baud.
Cable is test for continuity and isn’t broken.
Biqu seem to be asking me if there’s a short circuit between the data pins, but there isn’t on my skr1.4,
They didn’t ask for the screen to be tested but I did anyway, and the pins all have continuity between them apart from power and reset.
That doesn’t sound right to me but I don’t know how it’s supposed to work.
I’ll try uncommenting the sdcard_Connection lcd
If it’s even commented, I’m not sure I did anything other than what the instructions stated but it’s worth a try.

RavenWarrior, 2020-06-05:
Boom 💥 
Possible solution to my issue.
If none of the standard solutions like baud rate and port work for you, then this may be the problem.
Biqu reply after my Continuity readings. And the answer is yes to these.

Hello,
Is Rx continuity with all other pins?
Is Tx continuity with all other pins?
Is ground continuity with all other pins?
Can you give me a video about these three questions?
I will collect evidence for you, this is a short circuit of the display.
Send me your order number and delivery address by the way. I will forward you after sale

Yours sincerely,
BTT customer service

[7 more replies]