bigtreetech/BIGTREETECH-TFT35-V2.0 #13: Corrupted initial screen

kibinek, 2020-02-03:
I upgraded firmware of my BTT TFT35 V2.0 to version TFT35_V2.0.25.1 (Jan 11 2020), but when the initial screen loads up, it's corrupted (most of icons are missing and bottom middle area is filled with a "noise").
Please, do you suffer from the same issue? What could be the reason and fix?
[kibinek007.sweb.cz/3d_print/tft35v2_issue.jpg](url)

kibinek, 2020-02-06:
Well, currently I left Classic view and I use Unified view. Somehow it works :D

neonotthefirst, 2020-02-20:
Hey kibinek have the same issues, or even worse... updatet my 2.4 tft first and i was surprised how well the new UI looks and love the implemented functions. But after i flashed the new bootloader to the 3.5 v2, was tricky you need 3 hands :), the update was a mess. Any sugestions how to solve it?
[image]

Royalblu, 2020-03-13:
I have the same issue, did you guys get the above issues resolved?

kibinek, 2020-03-14:
As I mentioned in my previous message, my "solution" was to use Unified firmware instead of Classic.

neonotthefirst, 2020-03-19:
Solution cant be easier...
Bigtreetech updatet their TFT firmware page. Described how to copy the files to the sd card.
Just used my old compiled bin and copied the tft35 folder with the fonts and icons in it.
Solved...working fine:)