bigtreetech/BIGTREETECH-TFT35-V2.0 #23: The new FW for the display

radek8, 2021-04-11:
The new FW for the display is here.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

fjbold1, 2022-07-24:
> The new FW for the display is here. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

It contains a bigtree gd tft3.5 v2.0 .27 and a Biqu tft3.5 app1 v2.0 .27 which one?  also get unkown system.

radek8, 2022-07-24:
It depends on what HW you have.
If you don't know, take a picture of the display from both sides and send the photos.
Then I'll tell you which file...

fjbold1, 2022-07-24:
[image]
[image]

radek8, 2022-07-24:
Bad photos.
can you take a better picture of this so the text can be read?

[image]

[image]

fjbold1, 2022-07-24:
![image](https://user-images.githubusercontent.com/44207007/180662440-e057fb76-3705-44b9-ab8a-99973efff108
[image]
.jpg)

radek8, 2022-07-24:
Use:
BIQU_TFT35_APP1_V2.0.27.x.bin

fjbold1, 2022-07-24:
[image]

fjbold1, 2022-07-24:
Maybe I should use an older system file.

radek8, 2022-07-24:
What system are you using?
TFT is optimized for Marlin.

fjbold1, 2022-07-24:
the system file included in the software, there are two, one for reprap and the standard one,that I used. config.ini

radek8, 2022-07-24:
TFT updated fine.

What FW is your printer using?
Marlin, RepRap, Repetier, Smoothie, Klipper, Prusa... ?

fjbold1, 2022-07-24:
marlin, i never had a problem with updating.

radek8, 2022-07-24:
> I meant config file included in the software, there are two, one for reprap and the standard one,that I used. config.ini

Both are the same, they differ only in the preset default values, which you can edit before updating.

radek8, 2022-07-24:
Connect the printer,
if you have the correct port activating and the same speed of the serial port, the TFT should work.

fjbold1, 2022-07-24:
I'll try it on a machine on which I have a tft 2.4 working. I've had the tft3.5 for some time but never used it.

radek8, 2022-07-24:
Set the same serial port speed on the TFT35 as you have on the TFT24 and it will work as well

fjbold1, 2022-07-24:
no luck, I had another spare 2.4 and that works fine too. Set for 250000 , I can try recompiling at 115200, other than that, I'll experiment or just put it back in storage.(>:

fjbold1, 2022-07-25:
Finally success, I decided to upgrade my spare tft 2:4 v1.1, so I connected a skr 1.4 turbo to the usb port on the laptop and connected the tft 2.4. After the upgrade, there was a line on the info page with the marlin version that was used on the skr board that was not there before.  Next I connected the tft 3.5 and after a reset it connected to the board. I again ran the upgrade process on the tft 3.5 because I liked the look of the round icons, and after that it still connected to the 1.4 turbo. The info page still said “unknown system”, I then tried it on the machine that was working with another 2.4tft and it would not connect, the 2.4 that I just upgraded did work on that machine which is using a skr pro v1.

radek8, 2022-07-25:
You have many TFT displays and SKR boards.
I got lost in translation. :-)
So what is the result? Is everything working for you?

fjbold1, 2022-07-25:
I have an SKR mini v3.0 that I was planning to use with the 3.5 display, I will find out whenever I get around to it.